intel/zephyrjs-ide #43: [ocf explorer] extra res concatenated to path

poussa, 2017-02-14:
In the explore form, when you type /api/oic to the path field it goes out as /api/oic/res.

When you type in the correct form, /api/oic/res, it gors out as /api/oic/res/res

Nuke the extra res and use literally what is in the path.

siovene, 2017-02-14:
Fixed by 427dda86b8eff4244b263a350ca7846d575d7737

closed 2017-02-14 by siovene

siovene, 2017-02-14:
`/res` is the path used to discover resources. If I make it part of the server's `path`, then it will try to get single resource properties with `/api/oic/res/a/fan`, for instance, which is incorrect (correct would be `/api/oic/a/fan`.

So `/res` cannot be part of the path, unless I ugly remove it when querying a single resource.

It's better to solve what is basically a usability issue by simply removing a trailing `/res` from the path if the user puts one there.

poussa, 2017-02-14:
There are also the platform and device discoveries. So we have

```
/api/oic/res - resource discovery
/api/oic/d - device discovery
/api/oic/p - platform discovery
/api/oic/<resource-path>?di=<device-id> - resource retrieve
```

Plus the POST for updates.

siovene, 2017-02-14:
Yes but the service you're seeing right now can only real with resource discovery, i.e. it knows what to do with the response when it gets some resources.

We can later add device and platform discovery via separate subclasses of the OCF API service.

commit 3c73dab5 (2017-02-14): ocf-explorer: fix #43 again by sanitizing /res$ out of the path

The previous fix broke getting single resources because they don't start
with /res.