intel/zephyrjs-ide #45: Console windows will be locked

cuiyanx, 2017-02-14:
**Issue description**
when drag console windows up and release, this windows will be locked and pick up never

**Current behavior**
be locked and never pick up

**Steps to reproduce**
drag console windows up and release

**Zephyr.js IDE version:** `0.6.2 / 88b29683`

siovene, 2017-02-14:
I'm sorry but I don't understand the issue description... can you please elaborate?

I assume by "drag" you mean that you are resizing the Console window? Then you make it bigger and then release? And then what happens?

cuiyanx, 2017-02-14:
yes, resizing the Console window, this window will be bigger, and release it. you will see that window will be locked and never pick up.

siovene, 2017-02-14:
Can you please explain what you mean by "locked" and "never pick up"?

haoxli, 2017-02-14:
I think the screenshots will help to understand easier.
[image]
After drag the button up, it will disappear, the console window will be locked, and cannot be resized.
[image]

siovene, 2017-02-14:
Okay, thanks, I could reproduce it. I'll look into it.

closed 2017-02-16 by siovene

qiaojingx, 2017-02-17:
Verified with **JS IDE FOR ZEPHYR OS** / v0.8.3-alpha / be5c388
This issue is not reproducible.

commit 3f449415 (2017-02-16): Fix disappearing resize handle

Fixes: #45