intel/zephyrjs-ide #47: Command prompt not in the bottom of the console window

cuiyanx, 2017-02-15:
**Issue description**
When I using ashell command "run" to run my JS code, The command prompt(`acm>`) not in the bottom of the console window. But I input another ashell command, The command prompt(`acm>`) will be re-displayed in the bottom of the console window.

**Current behavior**
The command prompt(`acm>`) is in the center of my JS code's console logs.

**Expected behavior**
displayed in the bottom of the console window

**Steps to reproduce**
run JS code that will output a lot of console logs

**Zephyr.js IDE version:** `0.8.1 / c9027d6`

siovene, 2017-02-15:
Screenshot please? I don't see this.

I connected to a device and ran the following code on it:
```js
for (var i = 0; i < 100; i++) {
  console.log(i);
}
```

and the `acm>` prompt was always at the end of the output.

cuiyanx, 2017-02-15:
[image]

siovene, 2017-02-15:
Can you share the code that you used in the editor to generate that output?

cuiyanx, 2017-02-15:
test code:
[test-timer.js](https://github.com/01org/zephyr.js/blob/master/tests/test-timers.js)

cuiyanx, 2017-02-17:
@siovene Maybe this issue appears because of delay output console logs with zephyr.js timers. I test [test-pwm.js](https://github.com/01org/zephyr.js/blob/master/tests/test-pwm.js) like this issue.

siovene, 2017-05-08:
Is this still an issue?

qiaojingx, 2017-05-09:
Verified with version v0.13.2-alpha / 02c9ee1. This issue is still reproducible.

`acm>` appears in the result queue, not at the end of console log. Please refer to the screenshot.
[image]

But, `acm>` will print by pressing 'Enter' manually.

siovene, 2017-05-09:
Okay this is because `ashell` is sending data coming from the running up and we have no choice but to append it in the console because we cannot differentiate input coming because the user just typed something, or input coming that way.

@nagineni, does it make sense running an app on the device is asynchronous? I like it this way, but in that case we should mark the app's output so it goes somewhere else in the UI.

Yeah, we meed that protocol...

xiaoyu3x, 2017-09-21:
This issue is reproduced with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2018-04-17:
This issue isn't reproduced with commit 02a08ec, so closed it.

closed 2018-04-17 by brianjjones