intel/zephyrjs-ide #48: There are confusion about console logs and commands

cuiyanx, 2017-02-15:
**Issue description**
There are confusion about console logs and commands in the console window when I create more "TAB"s.

**Current behavior**
Display confusion

**Expected behavior**
Separately display each console windows.

**Steps to reproduce**
Create more "TAB"s to connect one USB device

**Zephyr.js IDE version:** `0.8.1 / c9027d6`

cuiyanx, 2017-02-17:
@siovene Maybe this issue is the same as #17 , I'm not really sure.

siovene, 2017-02-17:
@cuiyanx it's possible... I will confirm when I take a closer look, thanks!

siovene, 2017-05-02:
I can't reproduce this anymore in current master.

closed 2017-05-02 by siovene

qiuzhong, 2017-05-09:
I can't reproduce this issue either ([c013aaf](https://github.com/01org/zephyrjs-ide/commit/c013aaf852ffdae0b7fc0d3e5d9cea6c8e217bfb))