intel/zephyrjs-ide #56: Mobile - Responsive design

clagom, 2017-02-17:
**Issue description**
Resizing the window the design is not responsive
[image]

**Current behavior**
The layout doesn't fit to the resize of the window

**Expected behavior**
The layout should be responsive and react to the resize of the window with adjusting the UI
Example of the demo implementation visible here: http://launchbox-dev.ld.intel.com/editoride.php
[image]

**Steps to reproduce**
Resizing the window

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
JS IDE FOR ZEPHYR OS / v0.8.3-alpha / df57eba

clagom, 2017-03-20:
On mobile it looks very broken
[image]

siovene, 2017-03-20:
Does it even make sense to offer the editor on a mobile device? You couldn't do any work on them because they can't connect to the IOT devices.

I propose we make sure we are responsive for all other views, for people browsing and learning about the project, but we simply display a message "the IOT IDE does not support mobile devices. Please come back using a computer", or something like that, for the editor page.

commit ea9e97ba (2017-03-23): fix(editor): display 'unsupported' message on small devices

Closes #56
closed 2017-03-23 by siovene