intel/zephyrjs-ide #59: [UI]Resize windows not smoothly

qiaojingx, 2017-02-23:
**Issue description**
Resize windows not smoothly

**Current behavior**
Movement is not smoothly when try to enlarge editor window.

**Expected behavior**
Resize windows always smoothly.

**Steps to reproduce**
Resize windows by dragging the frame. Please refer to the pic.

**'JS IDE for Zephyr OS' version:** `v0.9.0-alpha / af066fe`

[image]

siovene, 2017-02-23:
@qiaojingx, is this behavior also reproducible when the "OCF EXPLORER" tab is selected on the right pane, instead of the CONSOLE tab?

If so, then it's because you need a version of `ng2-split-pane` that is not installed by default by the build system because I'm waiting for [my patch  there]https://github.com/wannabegeek/ng2-split-pane/commit/3cd6459fca68720ad532c0d3c335b6cef5b2ad5a) to be released.

A temporary work around would be to build `ng2-split-pane` yourself, as follows:

```bash
git clone https://github.com/wannabegeek/ng2-split-pane.git
cd ng2-split-pane
npm install
./node_modules/.bin/tsc -p src
cd /path/to/zephyrjs-ide
npm install /path/to/ng2-split-pane
```

Once `ng2-split-pane` is released on `npm` with my patch, then this won't be necessary anymore of course.

qiaojingx, 2017-02-24:
@siovene, this behavior is only reproducible when the "console" tab is selected.
And i can see "ng2-split-pane" folder is already there, under /zephyrjs-ide/node_modules/

siovene, 2017-02-24:
Alright then the issue will go away when I'm able to use the upcoming update of `ng2-split-pane`. I'll put some pressure upstream for a release :-)

closed 2017-03-13 by siovene

qiaojingx, 2017-03-14:
Verified with JS IDE FOR ZEPHYR OS / v0.10.1-alpha / 0725450. This issue is not reproducible.

commit 12876b98 (2017-03-13): chore(build): install ng2-split-pane 1.1.0 from npm

Closes: #59