intel/zephyrjs-ide #60: Compatibility message on not supported browsers

clagom, 2017-02-23:
**Issue description**
The message "Your browser does not support WebUSB" is not very visible, visually, and changes position according to the type of browser.

**Current behavior**
The message "Your browser does not support WebUSB" is not very visible, visually, and changes position according to the type of browser.
Safari:
[image]
Firefox:
[image]

**Expected behavior**
Consistent visual, and position cross browsers. we can use the error notification visual for this message, example:
[image]

**Steps to reproduce**
Open the editor in Safari, or Firefox

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`

siovene, 2017-02-23:
Would it be okay to have that as a sticky notification, i.e. one that doesn't disappear by itself but only if closed by the user?

clagom, 2017-02-23:
Yes I think so

commit e699fcf0 (2017-02-23): fix(editor): replace 'webusb not supported' message with notification

Fixes: #60
closed 2017-02-23 by siovene