intel/zephyrjs-ide #63: Code imported from GitHub is not up to date

qiaojingx, 2017-02-27:
**Issue description**
Code downloaded from GitHub in zepyhrIDE is not up to date

**Current behavior**
Code imported from GitHub is not up to date
After step 3: New added samples(UDPEchoServ4.js and UDPEchoServ6.js) are not found.
After step 5: Code for I2C.js is not updated.(Please check the pics)

**Expected behavior**
Code downloaded from GitHub is always up to date

**Steps to reproduce**

1. Click "GitHub" button in zephyrIDE and login with GitHub API token
2. Choose Repository as 'username'/zephyr.js and Choose Branch as master
3. Check code in samples folder
4. Click I2C.js

**'JS IDE for Zephyr OS' version:** ` v0.9.0-alpha / 0e5e943`

I2C.js downloaded from GitHub in zephyrIDE:
[image]

I2C.js on GitHub:

[image]

siovene, 2017-02-28:
Are you sure your fork of `zephyrjs` is up-to-date with the upstream on the `01org` organization on GitHub?

qiaojingx, 2017-03-01:
My mistake that i forgot to update my fork of `zephyr.js` on `01org` organization on GitHub.
Closed since it isn't reproducible after fork of `zephyr.js` is up-to-date with upstream.

closed 2017-03-01 by qiaojingx