intel/zephyrjs-ide #68: [UI] More than one tab display as highlight

qiaojingx, 2017-03-13:
**Issue description**

More than one tab display as highlight

**Current behavior**

[image]

**Expected behavior**

[image]

**Steps to reproduce**

1. Create TAB#2 and TAB#3 by click "new tab"
2. Activate TAB#1 by click "TAB#1"
3. Close TAB#2 by click “X” on “TAB#2”

**'JS IDE for Zephyr OS' version:** `v0.10.1-alpha / 5f1ccc5`

siovene, 2017-03-13:
Also reproducible by:

- Create Tab #2 
- Activate Tab #1 
- Create Tab #3

closed 2017-03-15 by siovene

qiaojingx, 2017-03-15:
Verified with JS IDE FOR ZEPHYR OS / v0.10.1-alpha / aadfc5e. This issue is not reproducible.

commit a754f446 (2017-03-15): fix(editor): fix issues with active tabs

Closes #68