intel/zephyrjs-ide #69: [UI] Button design is inconsistent

qiaojingx, 2017-03-14:
**Issue description**

Button design is inconsistent

**Current behavior**

1. Button 1 and 3 are inconsistent if click GitHub in different tabs.
2. Other buttons are more square than button 2.  Is that better to keep button consistent?
[image]
[image]

**Expected behavior**

Button design should be consistent.

**Steps to reproduce**

1. Create mutiple tabs. Click GitHub in TAB#1 and other tabs.

**'JS IDE for Zephyr OS' version:** `v0.10.1-alpha / 0725450`

clagom, 2017-03-14:
Will work on these mockups as discussed today

siovene, 2017-03-27:
For now I made all buttons round as the "New tab" and "Console" buttons. Will change later if a UI spec is released.

closed 2017-03-27 by siovene

clagom, 2017-03-27:
Sent the proposal for the GitHub UI, where we define also these UI elements for consistency

qiaojingx, 2017-03-29:
Verified with v0.12.1-alpha / 7789490. This issue is not reproducible.

commit a58e33a9 (2017-03-27): fix: conform button UI

Closes #69