intel/zephyrjs-ide #72: Action notifications, visual confirmations

clagom, 2017-03-14:
**Issue description**
Implementing the visual notifications for the actions like uploaded, running, saved, and so on

**Current behavior**
Not present

**Expected behavior**
[image]

[image]

**Steps to reproduce**
Current layout

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
/ v0.10.1-alpha / 5f1ccc5

siovene, 2017-03-27:
Do we need a notification after saving the tab? Seems a bit redundant as you need to explicitly save and confirm. I would leave such notifications for things happening indirectly, or tasks needing a confirmation. Does it make sense?

clagom, 2017-03-27:
Yes I agree, now that we have save and confirm this is no longer needed. Still keep it as UI elements for important information (green), error (the red we have), and the neutral grey for less important notifications.

siovene, 2017-05-02:
> Yes I agree, now that we have save and confirm this is no longer needed

Closing then.

closed 2017-05-02 by siovene