intel/zephyrjs-ide #73: Menu sidebar

clagom, 2017-03-14:
**Issue description**
Feature

**Expected behavior**
[image]

[image]

[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
/ v0.10.1-alpha / 5f1ccc5

clagom, 2017-03-14:
Will implement the mocukps with the content for the other items as well

siovene, 2017-03-23:
I closed this issue even tho some minor features are missing, like searching and sorting.
For the GitHub link in the sidebar, I'll wait for the UI spec.

commit 5374635f (2017-03-23): feat(editor): add sidebar with local files and examples

Saving files saves them to the browser's localstorage.
Examples are hardcoded.

Closes #73
closed 2017-03-23 by siovene