intel/zephyrjs-ide #78: [UI] Unable to resize again if console window enlarged to the max

qiaojingx, 2017-03-17:
**Issue description**

Unable to resize again if console window enlarged to the max

**Current behavior**

Unable to resize again if console window enlarged to the max
[image]

**Expected behavior**

Resize ok

**Steps to reproduce**

1. Enlarge console window to the max like the pic.
2. Try to resize console window again. 

**'JS IDE for Zephyr OS' version:** `v0.11.0-alpha / fe0b8be`

siovene, 2017-03-17:
I can't reproduce this, it seems to work fine for me. Are you able to provide more details?

qiaojingx, 2017-03-17:
[image]
[image]
Click and drag frame from position 1.  Click again to fix when move frame to position 2.
Then i will unable to resize it again unless to clear browser history.

siovene, 2017-03-17:
I still can't reproduce. Please look at the following video: https://youtu.be/kcYP9BfuZ80
Am I missing something?

qiaojingx, 2017-03-20:
This issue is reproduced by https://01org.github.io/zephyrjs-ide/#/editor. Sorry for missing this information. Could you please try again?
Here is my issue behavior:
https://youtu.be/AK2qXwvxGXg

closed 2017-03-20 by siovene

qiaojingx, 2017-03-21:
Verified with JS IDE FOR ZEPHYR OS / v0.11.2-alpha / b07fb43. This issue is not reproducible.

commit ad1cfe82 (2017-03-20): fix(editor): give split separator a high z-index

This prevents it from being behind the boardviewer when the boardviewer
is at size 0. Possibly a Chrome bug.

Closes #78