intel/zephyrjs-ide #83: [UI] Scrollbar of editor windows doesn’t follow frame closely

qiaojingx, 2017-03-22:
**Issue description**

Scrollbar of editor windows doesn’t follow frame closely

**Current behavior**

Scrollbar of editor windows doesn’t follow frame closely when click and drag the frame
[image]

**Expected behavior**

Scrollbar of editor windows always follow frame closely

**Steps to reproduce**

1. Input some code in editor windows make sure scrollbar show up.
2. Click and drag frame between editor and console windows.

**'JS IDE for Zephyr OS' version:** `v0.11.2-alpha / f5f131e`

siovene, 2017-03-24:
Did you take that screenshot after you stopped moving the resize handle, or at the same time while you were still moving it very fast?

qiaojingx, 2017-03-24:
At the same time while I am still moving it.

siovene, 2017-03-24:
Then I'm afraid there's nothing we can do about it. Besides, it's a very, very minor issue.

closed 2017-03-24 by siovene