intel/zephyrjs-ide #85: Footer modification

clagom, 2017-03-23:
**Issue description**
As discussed, footer modifications: remove the title, move the links aligned with the title in the header.

**Current behavior**
[image]

**Expected behavior**
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.8.3-alpha / eb16de3

siovene, 2017-03-23:
The title is already removed and the links are aligned. Do you want me to make them smaller and grey?

clagom, 2017-03-23:
If possible with the next release yes, please. I think they look a bit too visible now, not primary action (like the green we use). Font size can be the same, mainly a colour change, thanks

commit f195f0a5 (2017-03-24): fix(editor): change color of release/tag info in footer

Closes #85
closed 2017-03-24 by siovene