intel/zephyrjs-ide #94: JS file samples indentation

clagom, 2017-03-28:
**Issue description**
Currently with 4 space indentation, might be better to change to 2 space indentation 
CC @kenchris

**Current behavior**
Files with 4 space indentation

**Steps to reproduce**
Open a JS file from the example menu on the left sidebar

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.12.1-alpha / 592fc72

kenchris, 2017-03-28:
The way the JS language works, there is a lot of indentation steps and thus the JS community has pretty much settled on 2 space indentation as that will keep more content in view without too much horizontal scrolling. We should do that same in the samples

siovene, 2017-09-29:
I'm copying the files straight from the [Zephyr.js repo](https://github.com/01org/zephyr.js/tree/master/samples), and I'm not going to post-process them for something so minor. If you feel so strongly about the issue, please report against Zephr.js, thanks! Closing this one.

closed 2017-09-29 by siovene