intel/zephyrjs-ide #100: Sidebar - background click to close

clagom, 2017-03-29:
**Issue description**
A natural interaction is to click to the background out of the sidebar to close it

**Current behavior**
Close sidebar just by clicking on the X icon

**Expected behavior**
Click on the background / editor area to close sidebar

**Steps to reproduce**
Open a menu item from the sidebar

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.12.1-alpha / 592fc72

siovene, 2017-04-20:
PR submitted to relevant third party dependency: https://github.com/arkon/ng-sidebar/pull/69

closed 2017-04-22 by siovene

qiuzhong, 2017-04-24:
Verified with [9df718](https://github.com/01org/zephyrjs-ide/commit/9df7180a3dd59198601120778e2669e875272d5a), this issue is not reproducible.

commit 3124dfac (2017-04-22): fix(editor): close secondary sidebar when backdrop is clicked

Closes #100