intel/zephyrjs-ide #104: No output log in console most of the time after trying to connect A101

qiaojingx, 2017-04-01:
**Issue description**

No output log in console most of the time after trying to connect A101
**Current behavior**

No output log in console most of the time after trying to connect A101
[image]

**Expected behavior**

Commands list is expected each time when connect A101 after plugging or reseting
**Steps to reproduce**

1. Plug in A101 and click connect to establish connection
2. Reset A101 and click connect to establish connection

**'JS IDE for Zephyr OS' version:** `v0.13.0-alpha / 208a971`

qiuzhong, 2017-04-27:
Sometimes executing <code>sudo service ModemManager stop</code> can fix this but it doesn't always work.

siovene, 2017-04-27:
What about disabling ModemManager completely?

qiuzhong, 2017-04-27:
I'll disable it completely and see if this issue still occurs in future.

qiuzhong, 2017-04-28:
So far, I didn't encounter this issue after disabling the ModemManager with several host reboots.

@qiaojingx , please stop and disable the ModemManager service by
```
sudo systemctl stop ModemManager.service
sudo systemctl disable ModemManager.service
```
and see if this issue happens on your side?

nagineni, 2017-05-02:
I've not seen this behavior before. Make sure that the ModemManager is disabled before you connect the device to the Host.

qiaojingx, 2017-05-02:
Verified with commit a299ebd. This issue is not reproducible after disabling the ModemManager by rebooting.
Closed.

closed 2017-05-02 by qiaojingx