intel/zephyrjs-ide #107: [update-seed] Fail to deploy the dockerized version

haoxli, 2017-04-12:
**Issue description**
Build and run the dockerized version following the README guide successfully, but http://localhost:5555 cannot access in browser.

**Current behavior**
http://localhost:5555 can’t be reached.

**Expected behavior**
http://localhost:5555 can be opened.

**Steps to reproduce**
1. Set up docker environment in Ubuntu 16.04.
2. ```git clone https://github.com/01org/zephyrjs-ide```
3. ```cd zephyrjs-ide```
4. ```git check out update-seed```
5. ```sudo docker-compose build```
6. ```sudo  docker-compose up -d```

**'JS IDE for Zephyr OS' version:** `branch: update-seed / 208a971`

**Additional Information:**
Docker version: `17.03.1-ce`
Docker Compose version: `17.03.1-ce`

haoxli, 2017-04-12:
Check logs by `docker-compose logs`:
[image]

siovene, 2017-04-19:
Please verify against master, as the update-seed branch was merged.

Port is now 8000, not 5555.

closed 2017-04-19 by siovene

haoxli, 2017-04-20:
Verified on master(commit b78ea7f), this issue cannot reproduce.

commit 0c465a16 (2017-04-19): fix(build): fix docker images

Closes #107, closes #108