intel/zephyrjs-ide #108: [update-seed] Access error when start angular-seed-build-prod

haoxli, 2017-04-12:
**Issue description**
Permission denied when mkdir '/home/app/angular-seed/dist/prod' in **Production build and deployment**

**Current behavior**
Access error occurs when `docker-compose -f docker-compose.production.yml up angular-seed`

**Expected behavior**
`docker-compose -f docker-compose.production.yml up angular-seed` run successfully

**Steps to reproduce**
1. Set up docker environment in Ubuntu 16.04.
2. ```git clone https://github.com/01org/zephyrjs-ide```
3. ```cd zephyrjs-ide```
4. ```git check out update-seed```
5. ```sudo docker-compose -f docker-compose.production.yml build```
6. ```sudo docker-compose -f docker-compose.production.yml up angular-seed```
[image]

**'JS IDE for Zephyr OS' version:** `branch: update-seed / 208a971`

**Additional Information:**
Docker version: `17.03.1-ce`
Docker Compose version: `17.03.1-ce`

siovene, 2017-04-19:
Please verify against master, as the update-seed branch was merged.

closed 2017-04-19 by siovene

haoxli, 2017-04-20:
Verified on master(commit b78ea7f4bdb8350864f4ae1856c9676090bcda70), this issue cannot reproduce.

haoxli, 2017-04-21:
Hi, @siovene, this issue still can reproduce in master(commit b78ea7f4bdb8350864f4ae1856c9676090bcda70)

qiuzhong, 2017-04-25:
Verified with [9df7180](https://github.com/01org/zephyrjs-ide/commit/9df7180a3dd59198601120778e2669e875272d5a), this issue is not reproduced.

commit 0c465a16 (2017-04-19): fix(build): fix docker images

Closes #107, closes #108