intel/zephyrjs-ide #109: Inconsistent result when running TrafficLight.js in ashell and ide

qiaojingx, 2017-04-12:
**Issue description**

Inconsistent result when running TrafficLight.js in ashell and ide

**Current behavior**

1. Red led is on for 5s, other leds are off.
2. All led if off for 3s.
3. Green led next to the red one is on for 2s,  Red led is also turned on(5s).

**Expected behavior**

1. Red led is on for 5s, other leds are off.
2. All led if off for 3s.
3. Green led next to the red one is on for 2s, other leds are off.
4.  Red led is on for 5s, other leds are off.
All leds should be turn on singly.

**Steps to reproduce**

1. connect A101 to WebIDE.
2. Run [TrafficLight.js](https://github.com/01org/zephyr.js/blob/master/samples/TrafficLight.js) in it.
3. Check on board leds.

**'JS IDE for Zephyr OS' version:** `v0.13.0-alpha / c60c538`

This issue is not reproduced in ashell.

qiaojingx, 2017-04-13:
This issue is caused by the inconsistent test code between webide(sidebar/examples/Traffic Light) and GitHub([TrafficLight.js](https://github.com/01org/zephyr.js/blob/master/samples/TrafficLight.js)). Update issue description.

dbkinder, 2017-04-17:
BTW, this example won't work correctly in the WebIDE on a bare Arduino101 board.  As mentioned in https://github.com/01org/zephyr.js/blob/master/docs/a101_pins.md  "The flash filesystem uses SPI, so LED2 is not available when you use ashell or filesystem APIs."

grgustaf, 2017-05-09:
Yeah, this sample should really be dropped for that reason. Too bad, I like it when it works. :)

You might want to pull samples directly from ZJS repo at the time of publishing updates to the website version of IDE? Note: I have a pending change 01org/zephyr.js#1097 that will change the GPIO open syntax so many samples will change.

siovene, 2017-05-09:
Yeah @grgustaf, I think I'll simply reference those as a git submodule and copy them to the right place at build time.

siovene, 2017-05-18:
Resolved.

closed 2017-05-18 by siovene

qiuzhong, 2017-05-19:
Verified with [3ea94bb](https://github.com/01org/zephyrjs-ide/commit/3ea94bbb6b1153fb272df8e348ad86ef429ed9a6), due to #141 and PR #145, the TrafficLight.js is not needed any more and removed.