intel/zephyrjs-ide #114: "stop" command to running sensor app kills further communication

dbkinder, 2017-04-17:
**Issue description**
"stop" command issued on a running app (included below) causes lots of the same error message to be written out:
```
src/zjs_callbacks.c:511 zjs_call_callback():
[ERROR] callback 4284 does not exist
```

**Expected behavior**
No error messages written out.

**Steps to reproduce**
Here's the sample apps code:
```
// Light up the on-board LED0 (Red) if the board is moved/tilted
// otherwise, on-board LED1 (Green) if the board is still/flat

console.log("Starting Movement example...");

var gpio = require("gpio");
var pins = require("arduino101_pins");

// we'll use the on-board LEDs as an indicator
var ledR = gpio.open({pin: pins.LED0, direction: 'out'});
var ledG = gpio.open({pin: pins.LED1, direction: 'out'});

ledR.write(true);  // turn off the Red (fault) LED to get started
ledG.write(false); // turn on the Green (TX/RX) LED to get started

var change = 2.0;  // motion threshold to turn on the LED
var updateFrequency = 10 // for checking accelerometer (in Hz) 

var sensor = new Accelerometer({
    includeGravity: false, // true is not supported, will throw error
    frequency: updateFrequency
});

// define callback function for the accelerometer's onchange event

sensor.onchange = function () {
    // if x/y data is over the change threshold, Red otherwise Green         
    if ((sensor.x > change) || (sensor.x < -change) ||
        (sensor.y > change) || (sensor.y < -change)) {
        ledR.write(false);  ledG.write(true);
    } else {
        ledR.write(true);   ledG.write(false);
    }
};

sensor.start();

```
Connect and run this sample, wait for the green light, tilt the board a few times, and then enter "stop" in the console window.

**'JS IDE for Zephyr OS' version:** `v0.13.0-alpha / c60c538`

dbkinder, 2017-04-17:
Also, if you make a source change while the app is running (say to update the **change** threshold from 2.0 to 1.0 to make it more sensitive) and try to run the app again (without using a "stop" command in the console), the app hangs and the console becomes unresponsive -- requires board to be reset.

xiaoyu3x, 2017-09-20:
This issue isn't reproduced when using commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2017-09-21:
@dbkinder,  
Does this issue still reproduce on your side? Could you closed it if  it not reproduce.

brianjjones, 2018-04-17:
Is this still an issue? I am not able to reproduce it.

xiaoyu3x, 2018-04-18:
This issue isn't produced with commit 02a08ec

closed 2018-04-25 by brianjjones