intel/zephyrjs-ide #115: [UI]Board Explorer Section not able to be hidden or closed

qiuzhong, 2017-04-24:
**Issue description**
On master branch, the Board Explorer Section cannot be hidden or closed.

**Current behavior**
The Board Explorer Section is blank and it cannot be hidden or closed.

**Expected behavior**
This section should be able to be hidden or closed.

**Steps to reproduce**
1. <code>npm start</code> to launch the IDE
2. Go to http://localhost:8000/#/editor

**'JS IDE for Zephyr OS' version:** `X.Y.Z / 9df7180`
master branch

siovene, 2017-04-25:
@clagom, is it an intended behaviour that that user should be able to close the board explorer section? I didn't infer that from the UI specs.

clagom, 2017-04-27:
Sorry, I'm not sure to understand what the issue is here?
The board explorer section can be collapsed:
[image]
And it's never blank

siovene, 2017-04-27:
I think the reported wants the board explorer to be closed like the Console? @qiuzhong please elaborate.

qiuzhong, 2017-04-28:
Sorry for any misleading description.

I mean the rectangle with red border cannot be closed like the Console.
[image]

It's from http://localhost:8000/#/editor. There is no content in this area.
When the Console area grows to its maximum, it can cover the up area. But the board explorer can not be closed.

Is the board explorer are reserved on purpose and users are allowed to close it?

siovene, 2017-04-28:
This is the intended behavior.

closed 2017-04-28 by siovene