intel/zephyrjs-ide #117: Unable to differentiate multiple boards

qiuzhong, 2017-04-25:
**Issue description**
Unable to differentiate multiple boards when connecting them in multiple tabs

**Current behavior**
With multiple tabs and multiple boards, there are multiple instance of the boards but with the same name, users cannot differentiate them.

**Expected behavior**
When connecting the boards, each board should have a unique name for one tab.

**Steps to reproduce**
1. Connect 2 Arduino 101 boards to the host.
2. Launch the IDE and go to http://localhost:8000/#/editor.
3. Create 2 tabs, click "Connect" in any tab, you'll get this:

[image]

**'JS IDE for Zephyr OS' version:** `0.13.1-alpha / 9df7180`

siovene, 2017-04-25:
Unfortunately there is simply no way to differentiate them. I will give @nagineni a chance to comment before I close the issue, because quite frankly I can't swallow this fact either :-) But it seems that's just the way it is. At least on the client side all data is identical between two Arduino 101 devices flashed with Zephyr OS.

siovene, 2017-04-27:
As confirmed by @nagineni, this is simply not possible.

closed 2017-04-27 by siovene

qiuzhong, 2017-04-27:
It's acceptable. After connecting to the boards in order, by running different code between tabs and checking output, users can know how tabs are mapped to boards.