intel/zephyrjs-ide #118: [UX]Unable to unhighlight the tab titles via mouse clicking when renaming them

qiuzhong, 2017-04-25:
**Issue description**
During renaming tab titles, users cannot unhighlight the tab titles via mouse clicking.

**Current behavior**
When users click the edit button to rename the a tab title and the title is highlighted in a red background, they cannot unhighlight it via mouse clicking. 
Pressing the left or righ t arrow kyes can work well. Pressing any other keys will replace the title.

**Expected behavior**
Generally, when the users click the edit button of a tab, the tab title is highlighted in a red background.
Then clicking should unhighlight them and go to edit mode.

**Steps to reproduce**
1. Launch the IDE and go to http://localhost:8000/#/editor
2. Click the edit button of "Tab #1", now the tab title is highlighted in a red background.
3. In the tab area, try clicking to unhighlight the tab title and go to edit mode but fail.

[image]

**'JS IDE for Zephyr OS' version:** `0.13.1-alpha / 9df7180`

qiuzhong, 2017-05-27:
This issue cannot be reproduced any more.

closed 2017-05-27 by qiuzhong

siovene, 2017-05-27:
I spent a lot of time on this and I was unsuccessful, so now I'm very curious. Can you send me your current list of dependencies, so I can compare with mine? I haven't update in a while.

`npm ls > cat deps.txt`

Thanks!

siovene, 2017-05-27:
I just did that, actually, by saving my old dependency list, then doing `npm install` and comparing the new dependency list to the old. The `diff` doesn't seem to suggest anything related:

```
$ diff deps.txt deps-new.txt
448,450c448,450
< │ │ ├── UNMET DEPENDENCY colors@latest
< │ │ ├── UNMET DEPENDENCY cors@latest
< │ │ ├── UNMET DEPENDENCY event-stream@latest
---
> │ │ ├── colors@1.1.2
> │ │ ├── cors@2.8.3
> │ │ ├── event-stream@3.3.4
463,466c463,471
< │ │ ├── UNMET DEPENDENCY object-assign@latest
< │ │ ├── UNMET DEPENDENCY opn@latest
< │ │ ├── UNMET DEPENDENCY proxy-middleware@latest
< │ │ └── UNMET DEPENDENCY send@latest
---
> │ │ ├── object-assign@4.1.1
> │ │ ├─┬ opn@5.0.0
> │ │ │ └── is-wsl@1.1.0
> │ │ ├── proxy-middleware@0.15.0
> │ │ └─┬ send@0.15.3
> │ │   ├── debug@2.6.7
> │ │   ├── fresh@0.5.0
> │ │   ├── http-errors@1.6.1
> │ │   └── ms@2.0.0
474d478
< ├── cors@2.8.1 extraneous
1077d1080
< ├── proxy-middleware@0.15.0 extraneous
```

Also, this is still reproducible here with Chrome 58 on OSX. Did you also by any chance update your browser? Frankly this looks a lot like a browser issue.

qiuzhong, 2017-05-31:
I just did the test again on the same host.
* Ubuntu Linux 16.04 LTS
* node: v6.10.10; npm: v3.10.10
* Google Chrome: 58

1. I kept the directory when this issue was the file. By starting the IDE from this directory, I can still reproduce this issue.
I saved the dependency list as [deps_old.txt](https://github.com/01org/zephyrjs-ide/files/1040339/deps_old.txt).

2. Then I cloned a clean repository from github and reset the commit id to the one ([9df7180](https://github.com/01org/zephyrjs-ide/commit/9df7180a3dd59198601120778e2669e875272d5a)) when this issue is filed.
After running <code>npm install</code> and starting the IDE, this problem was gone.
I saved the dependency list as [deps_new.txt](https://github.com/01org/zephyrjs-ide/files/1040341/deps_new.txt).

That's why I said this issue cannot be reproduced any more. Because even we set back the code to [9df7180](https://github.com/01org/zephyrjs-ide/commit/9df7180a3dd59198601120778e2669e875272d5a), after npm installing, the dependency list will update automatically and differ from deps_old.txt.

By the way, I didn't try on OSX before. But I cannot reproduce this issue on macOS Sierra (10.12.5) with Chrome 58 today. Neither [9df7180](https://github.com/01org/zephyrjs-ide/commit/9df7180a3dd59198601120778e2669e875272d5a) nor the [latest status](https://github.com/01org/zephyrjs-ide/commit/b06c2d18e7de77c00d950800ef8179067458bd5f). Steps are the same as method 2 above.

I did update Chrome on Ubuntu but not on macOS before. It's hard to explain when we compare the 1 & 2 above if this is the a browser issue.
Did you use the same way when you reproduce it on OSX?

siovene, 2017-05-31:
Out of curiosity, can you test on Chrome 56? I think this is a browser bug that was fixed.

qiuzhong, 2017-05-31:
Sure. It will be a little troublesome to downgrade the Chrome browser. But I'll take a try.

qiuzhong, 2017-05-31:
Tested with Chrome 56 on Ubuntu (16.04) / Windows 10 / macOS  (10.12.5), I can't reproduce it.