intel/zephyrjs-ide #119: Personal Github respositories not loaded for picking files

qiuzhong, 2017-04-28:
**Issue description**
Personal Github respositories not loaded when login with Github Personal tokens.

**Current behavior**
All repositories of Otcshare orgnization are loaded.

**Expected behavior**
Your own Github repositories loaded for picking files.

**Steps to reproduce**
1. Start the IDE
2. Click Github of sidebar, input your personal Github API token
3. Otcshare respositories loaded, while they contain few files can be run in the IDE.

[image]

**'JS IDE for Zephyr OS' version:** `v0.13.2-alpha / 4aa60d3`

qiuzhong, 2017-04-28:
I am confused: why it needs my personal Github API token but loads Otcshare projects?

siovene, 2017-05-02:
When you create a GitHub AI token you get get to select "scopes":

[image]

Did you select the `repo` -> `public_repo` scope?

qiuzhong, 2017-05-02:
I selected all the scopes:

[image]

closed 2017-05-02 by siovene

qiuzhong, 2017-05-03:
Verified with [25d11ae](https://github.com/01org/zephyrjs-ide/commit/25d11ae14555e86513a69d180a56526db2fbb6f0), this issue is not reproducible. All personal repositories can be loaded for picking files now.

commit 25d11ae1 (2017-05-02): fix(github): get all repositories and not just 1 page

Closes #119