intel/zephyrjs-ide #122: Adding the logo mark to the website

clagom, 2017-05-02:
**Issue description**
Adding the logo mark (attached PNG image of the mark and examples)

[image]

[image]

[image]

**Current behavior**
Missing the logo mark

**Expected behavior**
Logo mark in the landing page and in the header of the website

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.13.2-alpha / 8050eef

clagom, 2017-05-02:
@siovene I attached a big sized logo mark for the retina displays and high resolution monitors.
Size in the landing page: 114 x 84 px
Size in the header: 38 x 28 px
I would remove the word "Hello" from the landing page when adding the logo, changing the margins and paddings like in the attached example.

clagom, 2017-05-02:
P.S. I see there's a misalignment now, we can fix that too:

[image]

siovene, 2017-05-03:
> P.S. I see there's a misalignment now between the title/mark and the call to action buttons below

I was playing a bit with that and it's problematic. If I align it so the two menu entries are center-aligned with the app name above, then the whole thing still looks off-alignment because of the `>` symbols.

If I center-align the call-for-action bar *as a whole*, then it looks off because the text would be too much to the let.

I propose we remove the `>` symbol and either color code the words themselves, or add a colored underline or something.

siovene, 2017-05-03:
PS: you should really only post one issue per report, because I need to close this after I add the logo to the app ;-)

clagom, 2017-05-03:
Right, opening a new issue for the alignment problem

commit aa3e9085 (2017-05-03): feat: add logomark

Closes #122
closed 2017-05-03 by siovene