intel/zephyrjs-ide #124: Misalignment in the landing page / buttons

clagom, 2017-05-03:
**Issue description**
P.S. I see there's a misalignment now between the title/mark and the call to action buttons below:
[image]

**Current behavior**
ul, menu, dir {
    display: block;
    list-style-type: disc;
    -webkit-margin-before: 1em;
    -webkit-margin-after: 1em;
    -webkit-margin-start: 0px;
    -webkit-margin-end: 0px;
    -webkit-padding-start: 40px; **< this is what cause the misalignment** 
}

**Expected behavior**
With this CSS modifications it seems fixed:
[image]

**Steps to reproduce**
Landing page

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.13.2-alpha / 8050eef

siovene, 2017-05-03:
As I was commenting in #122, I frankly don't like this solution. Even tho the menubar block is now center-aligned with the app's name above, my eyes seem to see the text and almost ignore the `>` symbols. So, if I focus, I can see that the alignment is correct, but at a first glance it seems that the menubar is too much to the left.

Can we get rid of the `>` symbols so the flow is more natural?

clagom, 2017-05-03:
We could put the > closer to the labels, so the block would be always centred but a bit more compact. I think as first step solution this might be ok, then we can think of redesigning the CTAs without > and with other visual solution. At the moment if we just remove the arrows > it looks weak and doesn't highlight the call to actions effectively.

siovene, 2017-05-03:
I still think it looks slanted. What about making them buttons with a background? That would settle the alignment because the eyes would look at the buttons as whole.

clagom, 2017-05-03:
I don't think this is a final solution, but if you prefer this version to the previous proposal then let's go on with buttons for now while we work on a better visual:

[image]

[image]

commit 208386e1 (2017-05-05): feat: improve alignment of home page navbar

Closes #124
closed 2017-05-05 by siovene