intel/zephyrjs-ide #128: [Fedora] Failed to detect WebUSB

qiuzhong, 2017-05-08:
**Issue description**
Failed to detect WebUSB on Fedora: 

**Current behavior**
No <strong>WebUSB detected</strong> alert when plug the device or reset it.

**Expected behavior**
When users plug the board or reset it, an <strong>WebUSB detected</strong> alert pops up.

**Steps to reproduce**
1. Follow the [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md) doc
2. Setting Chrome flags: #enable-webusb, #enable-experimental-web-platform-features enabled.
3. Start the IDE, and plug the board.

**'JS IDE for Zephyr OS' version:** `0.13.2-alpha / c013aaf`

** Other information:**
* Host: Fedora Workstation 25 (64bit)
* Chrome version on Fedora: 58.0.3029.96
* ModemManager service stop and disabled
* On Ubuntu, go to chrome://device-log:
  ```
  USB device added: path=/dev/bus/usb/001/008 vendor=32902 "Intel", product=63649 "WebUSB", serial="00.01", guid=36ea55cb-148a-472b-8fdf-5540107e8f79
  ``
* On Fedora, go to chrome://device-log:
  ```
  USB device added: path=/dev/bus/usb/002/060 vendor=32903 "Intel", product=2746 "GENUINO 101", serial="AE6776SQ609004J", guid=a4d19b60-9007-4290-becd-ba3031d0b6f4
  ```

nagineni, 2017-05-08:
@qiuzhong Did you create a udev rule as mentioned in the ashell docs? If not, could you please create a rule and try this again?

qiuzhong, 2017-05-08:
Yes. I created the <strong>/etc/udev/rules.d/99-arduino-101.rules</strong> and reloaded it. But it doesn't seem to work. 
```
$ ls /etc/udev/rules.d/99-arduino-101.rules 
/etc/udev/rules.d/99-arduino-101.rules
$ cat /etc/udev/rules.d/99-arduino-101.rules 
SUBSYSTEM=="usb", ATTR{idVendor}=="8086", MODE="0664", GROUP="plugdev"
```

nagineni, 2017-05-10:
@qiuzhong, looks like the 'plugdev' group doesn't exit in Fedora and that might be the reason for this issue. Try dropping group from the rule and see if it works or not.

You can try with this rule: 
SUBSYSTEM=="usb", ATTR{idVendor}=="8086", ATTR{idProduct}=="f8a1" , MODE="0666"

qiuzhong, 2017-05-10:
@nagineni , thanks!

I just tried the new rule you provided and it works well!

qiuzhong, 2017-05-10:
New rule 
<code>SUBSYSTEM=="usb", ATTR{idVendor}=="8086", ATTR{idProduct}=="f8a1" , MODE="0666"</code> fix the issue.

qiuzhong, 2017-05-10:
Would you update the [ashell](https://github.com/01org/zephyr.js/blob/master/docs/ashell.md) docs as well?

closed 2017-05-10 by qiuzhong

nagineni, 2017-05-10:
@qiuzhong Nice! Thanks for confirming.

I will update the instructions in ashell docs.