intel/zephyrjs-ide #129: [Windows] Failed to update webdriver

qiuzhong, 2017-05-08:
**Issue description**
Failed to update webdriver before running e2e tests

**Current behavior**
When running <code>npm run webdriver-update</code> to update webdriver, it fails.

**Expected behavior**
Run <code>npm run webdriver-update</code> without errors and all webdriver packages download.

**Steps to reproduce**
1. <code>git clone https://github.com/01org/zephyrjs-ide.git</code>
2. <code>cd zephyrjs-ide && npm install</code>
3. <code>npm install webdriver-manager && npm run webdriver-update</code>
```
C:\Workspace\verification\zephyrjs-ide>npm run webdriver-update

> zephyrjs-ide@0.13.2-alpha webdriver-update C:\Workspace\verification\zephyrjs-
ide
> node_modules/protractor/bin/webdriver-manager update

'node_modules' is not recognized as an internal or external command,
operable program or batch file.
...
```

**'JS IDE for Zephyr OS' version:** `0.13.2-alpha / c013aaf`

siovene, 2017-05-08:
Maybe try with PowerShell? The CI script works on appveyor: https://ci.appveyor.com/project/siovene/zephyrjs-ide/branch/master

qiuzhong, 2017-05-09:
This issue still remains in PowerShell on Windows:
```
PS C:\Workspace\verification\zephyrjs-ide> npm run webdriver-update

> zephyrjs-ide@0.13.2-alpha webdriver-update C:\Workspace\verification\zephyrjs-ide
> node_modules/protractor/bin/webdriver-manager update

'node_modules' is not recognized as an internal or external command,
operable program or batch file.
```

This issue happens when running <code>npm run webdriver-update</code> solely. But I didn't find this command was running from the CI log.

The [npm scripts webdriver-update](https://github.com/01org/zephyrjs-ide/blob/master/package.json#L42) is defined as:
```
"webdriver-update": "node_modules/protractor/bin/webdriver-manager update",
```
It works well on Linux but fails on Windows.

siovene, 2017-05-10:
Does the same happen with upstream angular-seed? https://github.com/mgechev/angular-seed

qiuzhong, 2017-05-11:
Tried with angular-seed.
This issue also exists in the upstream angular-seed.

siovene, 2017-05-11:
Then please report a bug there, that would be great.

qiuzhong, 2017-05-11:
New issue filed at upstream angular-seed, see [#1926](https://github.com/mgechev/angular-seed/issues/1926)

qiuzhong, 2017-05-24:
Upstream angular-seed had fixed this issue.

siovene, 2017-06-08:
Upstream merged.

closed 2017-06-08 by siovene