intel/zephyrjs-ide #130: [windows] Unable to claim device error

grgustaf, 2017-05-09:
**Issue description**
I tried to get the IDE working on my Windows 10 laptop today, but once I got some drivers installed such that WebUSB appears in the dialog, I always see the "Unable to claim device" error. I just saw now that if I have two copies of the IDE open on Linux and one is already connected I see this error from the second. So it's acting as if something else has already connected to it but I don't know what that would be.

**Current behavior**
Unable to claim device error.

**Expected behavior**
Able to claim device. :) Not sure if something on Windows 10 connects to it automatically or what? @brianjjones doesn't see this on Windows 8.1 I guess.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.13.2-alpha / 8050eef

poussa, 2017-05-09:
I have Win10 laptop (Dell) as well + Chrome 59. I was able to connect without any issues. I did not even install any additional USB drivers. We don't get the notification dialog on Windows so you need to launch the IDE manually. But after that everything worked.

poussa, 2017-05-09:
When I plug in the A101 board, I see 2 new USB devices in Device Manager

Other devices
  - WebUSB (with yellow warning icon)

Ports (COM &LPT)
  - USB Serial Device (COM3) ((with yellow warning icon)

nagineni, 2017-05-09:
@poussa I might have done the setup already on that Dell laptop :) I remember using that Dell when I was testing the WebUSB on windows. The landing page detection was disabled in Chrome due to some regressions on Windows and that's the reason for not seeing the notification.

poussa, 2017-05-09:
@nagineni yes, you are right.

xiaoyu3x, 2017-09-20:
This issue isn't reproduced when using commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2017-09-21:
@grgustaf ,
Does this issue still reproduce on your side? Could you closed it if  it not reproduce?

siovene, 2017-09-29:
Closing. @grgustaf please reopen if needed.

closed 2017-09-29 by siovene

grgustaf, 2017-09-29:
Sorry, yeah I think this was resolved.