intel/zephyrjs-ide #139: No way to connect a second device

qiuzhong, 2017-05-15:
**Issue description**
The IDE uses a global connection recently, a second device is not allowed to connect to the IDE anymore as the <strong>Connection</strong> button becomes <strong>Disconnect</strong>.

**Current behavior**
When the first device is connected to the IDE, there is no way to connect to connect to a new device.

**Expected behavior**
Multiple devices can be connected to the IDE.

**Steps to reproduce**
1. Launch the IDE, go to <code>/#/editor</code> page.
2. Connect the first device to the IDE, the <strong>Connection</strong> button becomes <strong>Disconnect</strong>.
3. Plug in a second device, an <strong>WebUSB detected</strong> pops up but there is no way to set up a new connection between the new device and IDE.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`0.14.0-alpha /  58bdc8e`

qiuzhong, 2017-05-15:
Is multiple devices connection support dropped completely?

siovene, 2017-05-15:
Yes, we don't support multiple devices at this time anymore.

closed 2017-05-15 by siovene