intel/zephyrjs-ide #140: Command palette submenu items  are truncated

qiuzhong, 2017-05-16:
**Issue description**
The Monaco editor <strong>Command palette</strong> submenu items are truncated: the starting two characters are truncated. 

**Current behavior**
Submenu items starting two characters are truncated

**Expected behavior**
Command palette submenu items are showing up well.

**Steps to reproduce**
1. Move the cursor to the Monaco editor section of the IDE
2. Right-click and choose <strong>Command palette</strong>, you get
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`0.14.0-alpha / 83a5d0e`

xiaoyu3x, 2017-09-21:
This issue is still reproduced with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2017-09-25:
This issue isn't reproduced with commit [9bdb7d7](https://github.com/01org/zephyrjs-ide/commit/9bdb7d728ff2fd82cf7bae6c54883dc4050855dd)

siovene, 2017-09-29:
Works fine for me too, closing.

closed 2017-09-29 by siovene