intel/zephyrjs-ide #141: [hackathon] Update Blink sample and remove TrafficLight

grgustaf, 2017-05-17:
**Issue description**
These don't seem to be in your repo so I can't submit a PR? But the Blink.js sample needs to be updated because the GPIO APIs have changed and it doesn't work as it is. The TrafficLight.js sample should be removed because one of the lights doesn't work in IDE mode so it doesn't make sense.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.15.0-alpha / efbd106

grgustaf, 2017-05-17:
You can find the latest version from ZJS [here](https://raw.githubusercontent.com/01org/zephyr.js/master/samples/arduino/basics/Blink.js).

siovene, 2017-05-17:
They're here:
https://github.com/01org/zephyrjs-ide/blob/master/src/client/app/pages/editor/example.service.ts

If you can submit a PR that would be a great favour, or if you're too busy I can take care of it too.

grgustaf, 2017-05-17:
Sorry, guess my git pull didn't work because I had some local change and I didn't notice. So I wasn't looking at the latest. Sure, I can do that.

grgustaf, 2017-05-17:
See PR #145

siovene, 2017-05-18:
PR merged, thanks!

closed 2017-05-18 by siovene