intel/zephyrjs-ide #142: [minor] Text unreadable on file selection panes

grgustaf, 2017-05-17:
**Issue description**
I think this is due to a recent update. When you click on Files or Examples, the text in the pane that opens listing the files is black on a dark grey background, and very hard to read.

I'm seeing this on Chrome 58 on Ubuntu 16.04.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.15.0-alpha / efbd106

siovene, 2017-05-17:
If I'm seeing this right, this is only on mouse-hover, right?

grgustaf, 2017-05-17:
You're right, I forgot that when I went to write it. :)

closed 2017-05-18 by siovene

qiuzhong, 2017-05-19:
Verified with [3ea94bb](https://github.com/01org/zephyrjs-ide/commit/3ea94bbb6b1153fb272df8e348ad86ef429ed9a6), this issue has been fixed. Here is the effect:

[image]

commit 751fd26f (2017-05-18): fix(editor): improve readability of files/examples in sidebar

Closes #142