intel/zephyrjs-ide #146: "echo off" appears sometimes in console

grgustaf, 2017-05-17:
**Issue description**
I don't think I was seeing this before today, but now sometimes I will see "echo off" in the Console and it seems like I'm not getting output. So I suspect something changed in the last 24 hours that caused this. I think the echo off support was already in before that, so it feels more like something in the IDE changed that is issuing the command in a different way. Basically, I wouldn't think the IDE would echo "echo off" and normally you don't see it.

It may or may not be related to more serious behavioral problem. If I hit run again though, it can sometimes get back out of it (I guess because it ends up calling "echo on").

I've seen it like 4x so I thought I'd report it just in case it rings a bell.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.15.0-alpha / efbd106

grgustaf, 2017-05-17:
BTW I think I can reproduce it most often if I hit "Run" pretty quickly after connecting the device, with a buffer of code already open.

siovene, 2017-05-17:
@nagineni can you take a look?

grgustaf, 2017-05-17:
I can reproduce it often if I reboot the device, connect as soon as it becomes available and hit Run quickly. (Which is not uncommon if I'm actively testing something.)

It displays echo off, and output is lost - I'm not sure whether the app is running or not. Then if I hit Run again, it still doesn't work. But if I hit Run a third time it seems to start working again. So it gets itself out of the bad state - but the normal user may have given up on that second try.

grgustaf, 2017-05-17:
OK, I'm surprised but that's why you test things I guess. I added an LED blink to my test script and I can see it's not actually running either first or second time I hit Run, only the third. (When this echo off thing happens.)

```js
var gpio = require('gpio');

varled = gpio.open("LED0");

var count = 0;
setInterval(function () {
    led.write(count % 2);
    count++;
    console.log("Item #" + count);
}, 250);
```

nagineni, 2017-05-17:
Looks like it is always reproducible if we send the transfer commands before ashell finishes the init parts.

siovene, 2017-05-18:
@nagineni, can we somehow detect from the console output when the `ashell` is ready, so we can set a flag?

nagineni, 2017-05-18:
We can look for acm> prompt and enable 'Run' only after we receive that from the ashell.

siovene, 2017-05-18:
Fixed by #150

closed 2017-05-18 by siovene

qiuzhong, 2017-05-19:
Verified with [3ea94bb](https://github.com/01org/zephyrjs-ide/commit/3ea94bbb6b1153fb272df8e348ad86ef429ed9a6), this issue is not reproducible.

nagineni, 2017-05-19:
@qiuzhong thanks for verification.