intel/zephyrjs-ide #147: [feature] Stop button would be useful

grgustaf, 2017-05-17:
**Issue description**
For apps with a lot of console output (like the Accelerometer and Gyroscope examples) it would be nice to have a Stop button, perhaps on the Console bar, or maybe Run / Running toggle like your Connect / Connected toggle.

If the user even realizes they can type stop in the Console window, it's hard to do it when you can't see what you're typing.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.15.0-alpha / efbd106

siovene, 2017-05-19:
@nagineni are we able to detect if something is running currently, so we can enable/disable or display either of the `Run` and `Stop` buttons? I don't quite like the idea of simply having them there as mappings to the corresponding `ashell` commands, unless that's our only option.

nagineni, 2017-05-19:
@siovene I don't see anything from ashell that tell us app is running on the device. I think we can set the flag when 'Run' is clicked and use that to enable the 'Stop' button for now.

siovene, 2017-05-19:
I don't like that: it means that when the program completes, the stop button will still be enabled and will never return to a disabled state. Better to never disable it at all if we really need it.

poussa, 2017-05-19:
I think this may cause more problems than it is solving. Can you put it on branch for now and we test it first there.

nagineni, 2017-05-19:
> I don't like that: it means that when the program completes, the stop button will still be enabled and will never return to a disabled state.

We can disable it when the 'Stop' button is clicked.

Anycase, I do not have a strong feeling whether we make the button toggle or not, but I like having a 'Stop' button in the IDE. :)

siovene, 2017-05-19:
Just pushed this to the `stop-button` branch, feel free to test.

```git
feat(editor): add STOP button

There is no enabling/disabling the run and stop buttons according to
whether we just begun running or we just stopped. I think we could do
something but it's pointless to try and get too clever only to be always
be missing some cases because there's really no "run status report"
coming from ashell. So let's leave it at the fact that the run and stop
buttons are simple mappings to the ashell console commands and be happy
with it.
```

qiuzhong, 2017-06-07:
Tested with the `Blink` example and the "Stop" button can function well.

Will this feature be available as well in the master branch?

siovene, 2017-06-08:
I think it's fine for me. I will merge it if no one raises objections.

commit ddc1a34d (2017-05-19): feat(editor): add STOP button

There is no enabling/disabling the run and stop buttons according to
whether we just begun running or we just stopped. I think we could do
something but it's pointless to try and get too clever only to be always
be missing some cases because there's really no "run status report"
coming from ashell. So let's leave it at the fact that the run and stop
buttons are simple mappings to the ashell console commands and be happy
with it.

Closes #147
closed 2017-06-22 by siovene