intel/zephyrjs-ide #155: atom-one-light.css is missing on 01org IDE site

qiuzhong, 2017-05-31:
**Issue description**
https://01org.github.io/zephyrjs-ide/app/pages/editor/components/board-explorer/board-viewer/bower_components/highlightjs/styles/atom-one-light.css is referred by https://01org.github.io/zephyrjs-ide/app/pages/editor/components/board-explorer/board-viewer/board-embedded.html. But it's missing.

**Current behavior**
404

**Expected behavior**
200

**Steps to reproduce**
1. Open chrome browser and go to https://01org.github.io/zephyrjs-ide/app/pages/editor/components/board-explorer/board-viewer/board-embedded.html
2. Go to the developers tools and click "Network". 

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.15.3-alpha / 6a05dff `

siovene, 2017-06-08:
No point in reporting issues related to the board viewer as it's going to be fully replaced with a different implementation.

closed 2017-06-08 by siovene

qiuzhong, 2017-06-14:
Has the new implementation been deployed on 01org site? Because when I open the [board-embedded.html](https://01org.github.io/zephyrjs-ide/app/pages/editor/components/board-explorer/board-viewer/board-embedded.html) on 01org site, the version has been updated (`v0.16.0-alpha / 9931d2d`) but the 404 error still exists.

siovene, 2017-06-14:
No, it will be replaced in the near future. I'm in the middle of rewriting it.