intel/zephyrjs-ide #158: Broken layout on the Board Explorer

clagom, 2017-05-31:
**Issue description**
The layout of the board explorer appears broken once you start interacting with it

**Current behavior**
Interaction breaks the layout of the board explorer

**Expected behavior**
I know we are going to rewrite the board explorer altogether, but at the moment is very broken. I wonder if there's a quick fix for the layout? Or we could prioritize the new board explorer.
[image]

The issue is not related to the size of the window or column.

**Steps to reproduce**
Click on the board

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.15.3-alpha / 6a05dff

clagom, 2017-06-05:
Might be better also to link/use this Genuino 101 board from the Board Explorer repositories directly: https://github.com/Board-Explorer/board-explorer/tree/master/boards/arduino_101

siovene, 2017-06-08:
No point in submitting issues for the board viewer as the whole thing is going to be replaced with a new implementation. (BTW this works fine for me on Chrome)

closed 2017-06-08 by siovene