intel/zephyrjs-ide #162: Some tests are not executed when running npm test on Windows 8.1

qiuzhong, 2017-06-12:
**Issue description**
For Ubuntu/Fedora/Windows 10, `npm test` will run all 34 tests and pass. But on Windows 8.1, only 23 tests are executed and pass. 

**Current behavior**
Some automatic tests are not executed on Windows 8.1 

**Expected behavior**
All 34 tests are executed when running `npm test`

**Steps to reproduce**
1. `git clone https://github.com/01org/zephyrjs-ide`
2. `cd zephyrjs-ide && npm install`
3. `npm test`

For the log, see the attachment. 
[npm_test_win8.1.txt](https://github.com/01org/zephyrjs-ide/files/1067459/npm_test_win8.1.txt)

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.16.0-alpha / 1c60c6f`
Chrome version: 59.0.3071.86

siovene, 2017-06-22:
Actually this happens on all platforms and the failure rate is not 100%.

qiuzhong, 2017-06-23:
Verified with `v0.18.2-alpha / 1400438` on the 4 platforms: ran the unit tests several times for each platform and this issue was not reproduced.