intel/zephyrjs-ide #163: Editor and home component e2e tests failed

qiuzhong, 2017-06-12:
**Issue description**
All the editor and home component e2e tests failed.

**Current behavior**
The error log shows the element cannot be located any more when running editor and home component e2e tests.

**Expected behavior**
All the e2e tests pass

**Steps to reproduce**
1. `git clone https://github.com/01org/zephyrjs-ide`
2. `cd zephyrjs-ide && npm install`
3. In terminal 1, `npm run serve.e2e`
4. In terminal 2, `npm run e2e`

See the detailed e2e error log:
[e2e_test_log.txt](https://github.com/01org/zephyrjs-ide/files/1067779/e2e_test_log.txt)

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.16.0-alpha / 1c60c6f`
Chrome version: 59.0.3071.86

siovene, 2017-06-12:
- Does this have a 100% failure rate? Because CI on Appveyor is fine: https://ci.appveyor.com/project/siovene/zephyrjs-ide/branch/master
- What about simply `npm run e2e.ci`?

qiuzhong, 2017-06-13:
1. Yes. It happend on Ubuntu/Fedora/Windows 8.1/Windows 10.
2. Simply `npm run e2e.ci` can get all e2e tests passed. It seems that `npm run e2e.ci` builds **prod** e2e test code while `npm run serve.e2e` builds **dev** one.

closed 2017-06-22 by siovene

qiuzhong, 2017-06-22:
Verified with `v0.17.0-alpha / 292ef61`. This is not reproducible: all the e2e tests can get through.

commit 5f5fdc78 (2017-06-22): fix(ci): wait for main component to be visible before each test

Closes #163