intel/zephyrjs-ide #165: Settings options - UI

clagom, 2017-06-13:
**Issue description**
Sent screens:
[image]
[image]
Settings icon from Font-Awesome

**Current behavior**
none

**Expected behavior**
Mock-up

**Steps to reproduce**
-

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.15.3-alpha / 6a05dff

siovene, 2017-07-18:
Added initial implementation with "Font size" setting. More settings will be added in time.

closed 2017-07-18 by siovene

xiaoyu3x, 2017-07-25:
Verified with v0.18.3-alpha / c803caa, "Font size" and "Line numbers" function work well.

commit f0236e12 (2017-07-18): feat(editor): add settings UI with fontSize option

Closes #165