intel/zephyrjs-ide #166: Device disconnected on Windows when running large size JS files

qiuzhong, 2017-06-14:
**Issue description**
The Arduino 101 device is very concerned about the size of uploading JS file. 
Currently, the largest size of the example is `Sensor BLE Demo.js` (123 lines, 3.0KB) and it can be executed via the IDE successfully. 
JS files are larger than that (133 lines console.log statements, 3.96KB in my experience) are easy to fail to run or to save to the device (It hangs).

I encountered the disconnection on Windows when running a large size JS file (360 lines, about 10.7KB)
In fact the largest [zephyr.js sample](https://github.com/01org/zephyr.js/blob/master/samples/HeartRateDemo.js) is about 11.6KB

**Current behavior**
Transferring error leads to device disconnection.

**Expected behavior**
Not sure.

**Steps to reproduce**
1. Generate a JS file, simply add 360 lines `console.log('hello, tab x')` from 1 to 360
2. Copy the JS file to the IDEand click **Run**
3. On Linux, it usually hangs. On Windows, it might pop up two alerts:
[image]

  and the device is disconnected soon.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.16.0-alpha / 1c60c6f`

brianjjones, 2017-06-16:
This actually isn't an issue with windows only, it happens on all platforms.  See #168.  Basically the IDE is sending data faster than the device can handle it, and on large files it causes the device to hang.

brianjjones, 2017-07-07:
This will fix the issue https://github.com/01org/zephyrjs-ide/pull/175

closed 2017-08-04 by siovene

wanghongjuan, 2017-08-07:
Verified this issue using test file about 8642 byte with commit 4f773db, the result is unstable, it can be passed at first time, the fifth time is failed. Consider the memory of Arduino 101, I think this is acceptable.

commit 991973eb (2017-08-04): feat(editor): add setting to throttle device transmission

Enabling this setting will make the delay in the webusb transfer
loop after transferring every 20 lines to prevent the UART from
overflowing.

Closes #166