intel/zephyrjs-ide #167: OCF module is not supported

wanghongjuan, 2017-06-16:
**Issue description**
I think all features which supported in zephyr.js should be supported in ide. I want to create a demo using [ocf](https://github.com/01org/zephyr.js/blob/master/docs/ocf.md) module and try to edit the code via ide, unfortunately, the module is not supported in ide, but can be required in zephyr.js.

**Current behavior**
Show message as below:
```
acm> In function require():
NotSupportedError: module not found: ''
```

**Expected behavior**
The ocf module should can be required

**Steps to reproduce**
This is my test code:
```
var ocf = require("ocf");
console.log('ocf is supported')
```

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
v0.16.0-alpha/c78adbb

siovene, 2017-06-16:
I think this has nothing to do with the IDE,  but it's related to the build of your device and what you put in there? cc @nagineni

nagineni, 2017-06-16:
Ashell, OCF, BLE and IP stack doesn't fit on A101 and that's the reason we are skipping OCF in the ide builds.

closed 2017-06-16 by siovene

wanghongjuan, 2017-06-20:
Thanks for your clarification. I didn't find there is documentation which said these features are disabled in IDE. Should this point be record in documentation?