intel/zephyrjs-ide #168: Need to slow the send rate on the run() function

brianjjones, 2017-06-16:
**Issue description**
Currently the IDE sends the JavaScript file as fast as it can to the UART.  The problem is that on larger files (150+ lines) this fills up the available memory for ashell before it can start writing it to the filesystem.  When this happens it causes ashell, and subsequently ZJS, to hang.

**Current behavior**
IDE is sending data too fast to the device
**Expected behavior**
IDE slows data send speed.

**Steps to reproduce**
Put a large (150+ line) JS file in the editor and hit run.  

In the long run this will be fixed by the ashell / IDE refactoring.  Which will set up a true send / ack protocol, preventing either from sending too much data before the other is ready.  But for now, slowing the speed at which the IDE sends lines of code to the device will prevent it from causing the device to overflow its memory.

brianjjones, 2017-06-16:
I'll also be looking to see about adding code to UART to prevent it from overflowing the memory on the device.

siovene, 2017-06-22:
@nagineni can you take a look?

siovene, 2017-06-22:
Actually this is a duplicate of #166, so let's track that one.

closed 2017-06-22 by siovene