intel/zephyrjs-ide #170: Information of some components is missing

qiuzhong, 2017-06-26:
**Issue description**
Information of some board components is missing. Their detailed information is not complete when clicking them. Either component or description or both values are empty. This exists in the **minnowboard_turbot_B_41** / **quark_mcu_dev_kit_c1000** / **quark_mcu_dev_kit_d2000** boards so far.

**Current behavior**
Some component information is not complete.

**Expected behavior**
All clickable components information is complete.

**Steps to reproduce**
1. Launch IDE
2. Select **Minnowboard** / **C1000** / **D2000** board respectively.
3. Check the components, click them and check the detailed information.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.18.2-alpha / 1400438`

siovene, 2017-06-26:
@poussa, can we get the data files (SVG, JSON, MD docs) from the Board Explorer project to be:

1. in a dedicated repository
2. updated and maintained?

closed 2017-08-01 by siovene

qiuzhong, 2017-08-03:
Some information is still missing. For example, Select **C1000** board and click **BLE**:
[image]

commit 5fde8bd2 (2017-08-01): chore(board-viewer): update boards

This brings the board definitions up-to-date with the BoardExplorer
project, and adds the UP board as well.

Closes #177, closes #178, closes #170