intel/zephyrjs-ide #171: Failed to stop the Gyroscrope example

qiuzhong, 2017-06-27:
**Issue description**
I tried to run and stop the examples one by one (Except `Pattern Matching` due to upstream [zephyr.js bug #1262](https://github.com/01org/zephyr.js/issues/1262)). When it came to the `Gyroscope` example, the **Stop** button didn't function.

**Current behavior**
After trying to stop the `Gyroscope` example, the IDE cannot behave well.

**Expected behavior**
Running and stopping the examples one by one can get through.

**Steps to reproduce**
1. Launch the IDE and connect to the board device.
2. Open the `Accelerometer` example from the sidebar. Click the "Run" button, wait for a while (3~5s) then click the "Stop" button.
3. Repeat step 2 until for example `BLE` and `Blink`. Now everything is fine.
4. Repeat step 2 for example `Gyroscope`. Running is still fine but not for stop. There's not `stop` in console output.
5. Now for example `Sensor BLE Demo`, neither "Run" or "Stop" can function.

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.18.2-alpha / 1400438`

qiuzhong, 2017-06-27:
Actually this happens not just for the `Gyroscope` example. It can easily happen when **Run-Stop** for more than 3 examples. After that, "Run" and "Stop" buttons don't work (maybe the device is too busy?)

xiaoyu3x, 2017-09-21:
This issue isn't reproduced  with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2018-04-18:
This issue is still produced with commit 02a08ec, so closed it.

closed 2018-04-19 by qiuzhong