intel/zephyrjs-ide #172: Device connected but "Run" and "Stop" are disabled

qiuzhong, 2017-06-27:
**Issue description**
Sometimes the IDE connection button shows the device is connected. But the "Run" and "Stop" buttons are both disabled. The Console output shows the `Connection established` but is not in the `acm>` interactive mode.

**Current behavior**
Device is connected. "Run" and "Stop" are both disabled.
Meanwhile, you cannot input any text in the console output.

**Expected behavior**
Not sure.

**Steps to reproduce**
1. **Run - Stop** the examples `Accelerometer` / `BLE` / `Blink` / `Gyroscope`.
2. Disconnect and reconnect to the device.
[image]

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.18.2-alpha / 1400438`

xiaoyu3x, 2017-09-21:
This issue isn't reproduced  with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2017-09-25:
This issue is reproduced with commit [9bdb7d7](https://github.com/01org/zephyrjs-ide/commit/9bdb7d728ff2fd82cf7bae6c54883dc4050855dd)

xiaoyu3x, 2017-09-25:
Recurrrence probability is 9/10 times if According to these steps:
1. Launch IDE and connect device.
2. Put 'Accelerometer' from Examples into device.
3. `boot  Accelerometer `
4. `reboot`
5. Reconnect device

xiaoyu3x, 2018-04-17:
@qiuzhong 
This issue isn't reproduced with commit 02a08ec, so closed it, please

closed 2018-04-17 by brianjjones