intel/zephyrjs-ide #173: Need to update Pattern Matching example code

qiuzhong, 2017-06-28:
**Issue description**
The `Pattern Matching` example requires `pme` module. However, from [zephyr.js bug 1262](https://github.com/01org/zephyr.js/issues/1262):

> PME is now dropped from ide because including it no longer fit any more, so it's intended.

The `Pattern Matching` code needs to update.

**Current behavior**
Failed to run `Pattern Matching` example.

```
acm> Curie Pattern Matching Engine test...
In function require():
NotSupportedError: module not found: 'pme'
```

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`
`v0.18.2-alpha / 1400438`

xiaoyu3x, 2017-09-21:
This issue is still  reproduced  with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2017-09-29:
The Pattern Matching example requires pme module. However, from zephyr.js bug 1262:

`PME is now dropped from ide because including it no longer fit any more, so it's intended.`

Please, delete 'Pattern Matching' from examples,
in addition, can provide a doc to list that module are removed from ashell/IDE test.

commit ff420550 (2017-09-29): fix(examples): remove PME example

Closes #173
closed 2017-09-29 by siovene