intel/zephyrjs-ide #179: Compilation errors when launching

qiuzhong, 2017-07-31:
**Issue description**

**Current behavior**

**Expected behavior**

**Steps to reproduce**

**'JS IDE for Zephyr OS' version:** `X.Y.Z / abcd1234`

xiaoyu3x, 2017-09-21:
This issue is still  reproduced  with commit [c5b35be](https://github.com/01org/zephyrjs-ide/commit/c5b35be17d8cc63de91c5c2da4d28308617bf882)

xiaoyu3x, 2017-09-25:
This issue isn't reproduced with commit [9bdb7d7](https://github.com/01org/zephyrjs-ide/commit/9bdb7d728ff2fd82cf7bae6c54883dc4050855dd)

xiaoyu3x, 2017-10-10:
hi, @qiuzhong 
This issue isn't produced with commit [3ca5e23](https://github.com/01org/zephyrjs-ide/commit/3ca5e23effd1ef9dddcfa49a963c660fd225cf0a), please closed it.

siovene, 2017-10-10:
Can't reproduce it either, closing.

closed 2017-10-10 by siovene